PrintM3D/Promega #2: Head thermistor fault.

benburch93, 2018-07-08:
I have an issue with the connector on the head thermistor.

I was getting high readings, and re-seated cables a few times and never got it down to ambient.

Then it went to 2000 degrees and fault.

I took the back off the cable box and disconnected the thermistor an noticed that one of the male pins was pushed back and I fixed that with a pick.  But I was still getting a high reading.  Then I noticed the one of the wires into the female connector was basically floating in its socket.  It appears that it was crimped with some of the fabric and this prevented a solid connection.  When I pull back the fabric and push the bare wire in, we then read ambient, but it does not remain that way unless held in place.

I am going to attempt a temporary repair with metal foil pushed in behind the wire, but I do not expect that t last long term.

Can I please get a new thermistor sent out?

David B. Ben Burch
668 Forest Ave
Elgin IL 60120-4461

Thank you!

benburch93, 2018-07-08:
PS - tried putting this onto air table and I couldn't get it to go...

PrintM3D, 2018-07-09:
noted, removed your personal info.

daniel-farina, 2018-07-31:
Closing this issue as we are aware of it and it's currently on the Airtable. We will transition all issues to Github eventually.

closed 2018-07-31 by daniel-farina