PrintM3D/Promega #11: part 2?

jbolene, 2018-08-28:
it is the end of August, part 2 was supposed to be posted

sohjsolwin, 2018-09-06:
I agree with this. I'm looking forward to more information included in the second half of the promega update. Will this be coming soon or is there a new estimated time frame?

ericl-m3d, 2019-04-03:
Hello,

Please log ONLY issues pertinent to software bugs or operational anomalies (suspected to be software-related).
Any other type of issues logged will be not be addressed and will be deleted.

If you are looking for support, feel free to look through our ever-improving documentation for quick help. 

- [M3D's Official Troubleshooting & Self-Help Guide](https://coda.io/d/M3D-Official-Troubleshooting-Self-Help-Guide_dzE73kMbIAL/Introduction_sucuq#_lulpw)
- [Promega Guides & Instructions](https://promega.printm3d.com/)

Our support ticket is available at https://store.printm3d.com/pages/support.

If you'd like to voice your input on improving our documentation, visit: 
- M3D's Official Troubleshooting & Self-Help Guide [Contribution & Suggestion Page](https://coda.io/d/M3D-Official-Troubleshooting-Self-Help-Guide_dzE73kMbIAL/Coda-Edits-Requests-Form_suKF6#_luq0Y)
- Promega Guides & Instructions [Contribution & Suggestion Page](https://promega.printm3d.com/#contributing)

-M3D Team

closed 2019-04-03 by ericl-m3d