PrintM3D/Promega #14: quad head status

jbolene, 2018-08-28:
the kickstarter surveys were supposed to go out this week
what is the status?
when are they shipping?
are they faster than the compound?
where are the mounts?

michaelarmani, 2018-08-31:
surveys are going out as we speak.
Mounts are being internally developed / under testing currently
they are shipping over the course of the next 1-2 weeks.
They seem to be able to extrude quite a bit faster than compound, probably owing to a variety of factors.  45 mm/s  0.2mm layers 0.4mm width = 3.6mm^3/s for PLA is where we were on prototype versions which were set very conservatively.

jbolene, 2018-09-03:
I am backer #1 group #1, seems if surveys were going out that I would have one

sohjsolwin, 2018-09-06:
@jbolene are you referring to the crane quad backing on FFL, the QuadFusion print head only backing on FFL, or the QuadFusion backing on Kickstarter?

closed 2019-04-03 by ericl-m3d