PrintM3D/Promega #8: changelog

jbolene, 2018-08-28:
Why are there no descriptions of what changed and why?
There should also be a before and after pic.
How are we to know what version we have?
Since there are no reasons for change we have no idea what the reasons are and if we need to make those changes also.

jbolene, 2018-08-28:
https://promega.printm3d.com/changelog

nothing to tell me what changed and why, no pictures, no descriptions

sohjsolwin, 2018-09-06:
Very seldom do change logs include a "why" for what was changed. Generally, they will include a link to the issue that the change resolved, at best. As more issues are included in these github repos, links like that will be possible. I agree that including some pictures or something next to the STL download files would be beneficial for users to visibly see the file. Alternatively, uploading these files to Thingiverse or a similar site that automatically generates thumbnails for them would be helpful and fulfill the same purpose.

sohjsolwin, 2018-09-06:
Linking to [Promega Docs issue](https://github.com/PrintM3D/Promega-Docs/issues/14)

ericl-m3d, 2019-04-03:
Hello,

Please log ONLY issues pertinent to software bugs or operational anomalies (suspected to be software-related).
Any other type of issues logged will be not be addressed and will be deleted.

If you are looking for support, feel free to look through our ever-improving documentation for quick help. 

- [M3D's Official Troubleshooting & Self-Help Guide](https://coda.io/d/M3D-Official-Troubleshooting-Self-Help-Guide_dzE73kMbIAL/Introduction_sucuq#_lulpw)
- [Promega Guides & Instructions](https://promega.printm3d.com/)

Our support ticket is available at https://store.printm3d.com/pages/support.

If you'd like to voice your input on improving our documentation, visit: 
- M3D's Official Troubleshooting & Self-Help Guide [Contribution & Suggestion Page](https://coda.io/d/M3D-Official-Troubleshooting-Self-Help-Guide_dzE73kMbIAL/Coda-Edits-Requests-Form_suKF6#_luq0Y)
- Promega Guides & Instructions [Contribution & Suggestion Page](https://promega.printm3d.com/#contributing)

-M3D Team

NOTE: Thank you for your suggestion and input.

closed 2019-04-03 by ericl-m3d