PrintM3D/Promega #19: Plastic Insert Wear

tirelessirony, 2018-08-31:
[image]

In trying to track down the cause of extruder gear slipping on PLA prints I started to examine the walle and extruder gears. I had reassembled the K'tana once without the black insert, so I knew that without it 1.75 filament slides forward in the 'feed channel' and isn't pushed effectively by the extruder gear at all. I hypothesized wear on the black insert might account for play developing into skipping.

To test it, I put calipers on the black insert from my K'tana head to check the difference between the unused side and the side which had printed using ABS-R. here's the results of the thickness. The piece of plastic along the edge (the part that sits in the feed channel) tapers toward the top and is thickest where it butts up against the ptfe tube:
                             Left side (used for ~200hrs)/ Right side (nearly unused entirely)
thinnest part :      2.14mm       /   2.21mm
center             :      2.43mm      /    2.56mm
thickest part :      2.79mm      /    2.93mm 

if you look at the side that has ~200hrs of use, there is a small, shiny, filament sized and shaped groove appearing (see photo). After private discussions with atleast one other user w/ 1000 hrs of use and seeing substantially more wear, I present y'all with the theory wear on the insert could be contributing to direct drive gear skipping that presents as underextrusion. Thanks for your time.

J-wMueller, 2018-08-31:
Same here, can post pictures of two different inserts with wear.

michaelarmani, 2018-08-31:
Thanks, we had planned to increase the longevity to about ~10000 hrs using nylon.

jbolene, 2018-09-03:
why not use carbon petg?  abrasiveness?
when will the new ones be available and will they be shipped to existing customers?

closed 2019-04-03 by ericl-m3d

Turtle49, 2020-11-10:
Same happened to me. I am waiting for a Nylon insert 3D printed by a commercial printing service. Teflon/PTFE may be the best material to use, though can't be printed.