PrintM3D/Promega #21: Not getting any extrusion at all.

hallambaker, 2018-09-09:


hallambaker, 2018-09-10:
OK, have dispensed with the PTFE tubes and it is now working.

Next problem: getting PLA to stick to the bed.

sohjsolwin, 2018-11-01:
60mm/s is way too high for most filaments for an open air extrude. You will also quickly run into issue with PLA and not using the PTFE tubes. PLA does not have the necessary lubricants in the material to not burn and stick to the metal. If using the Duet Web Controller, 5-15mm/s is generally the max you want to use through that UI when testing.

ericl-m3d, 2019-04-03:
Hello hallambaker,

Please log ONLY issues pertinent to software bugs or operational anomalies (suspected to be software-related).
Any other type of issues logged will be not be addressed and will be deleted.

If you are looking for support, feel free to look through our ever-improving documentation for quick help. 

- [M3D's Official Troubleshooting & Self-Help Guide](https://coda.io/d/M3D-Official-Troubleshooting-Self-Help-Guide_dzE73kMbIAL/Introduction_sucuq#_lulpw)
- [Promega Guides & Instructions](https://promega.printm3d.com/)

Our support ticket is available at https://store.printm3d.com/pages/support.

If you'd like to voice your input on improving our documentation, visit: 
- M3D's Official Troubleshooting & Self-Help Guide [Contribution & Suggestion Page](https://coda.io/d/M3D-Official-Troubleshooting-Self-Help-Guide_dzE73kMbIAL/Coda-Edits-Requests-Form_suKF6#_luq0Y)
- Promega Guides & Instructions [Contribution & Suggestion Page](https://promega.printm3d.com/#contributing)

-M3D Team

NOTE: This [sub-section](https://coda.io/d/M3D-Official-Troubleshooting-Self-Help-Guide_dzE73kMbIAL/Promega-Troubleshooting_suSw-#_lurJI) (within the Troubleshooting & Self-Help Guide) should start you off in the right direction, before submitting a support ticket (if needed).

closed 2019-04-03 by ericl-m3d