PrintM3D/Promega #23: Two parts laying on floor of printer when unboxed

jackmartintx, 2018-09-18:
When the printer was unboxed there was a screw and what looks like a cable guide- black elbow - used on other corners - on the bottom.  I'm not quite sure where they go or if they are just "extra".  The bed is also titlted down to the back left corner.

jackmartintx, 2018-09-18:
Sometimes you just have to write it for the lightbulb to go off.  This is a guide for the back left cable "mast".  It is broken.  I believe that it snaps in place.  The back "snap in" is broken off.  I am basing this on the same part just below the print bed.  on basing this on the fact that the front mast has two of these sliders.  The back only the one.  I can tell from the other three that this one is broken.

1.  How do I get a new one?
2.  Do I stop all assembly and later operation until I have this replaced.
3.  I have not figured out where the screw goes.  
[image]

ericl-m3d, 2019-04-03:
Hello jackmartintx,

Please log ONLY issues pertinent to software bugs or operational anomalies (suspected to be software-related).
Any other type of issues logged will be not be addressed and will be deleted.

If you are looking for support, feel free to look through our ever-improving documentation for quick help. 

- [M3D's Official Troubleshooting & Self-Help Guide](https://coda.io/d/M3D-Official-Troubleshooting-Self-Help-Guide_dzE73kMbIAL/Introduction_sucuq#_lulpw)
- [Promega Guides & Instructions](https://promega.printm3d.com/)

Our support ticket is available at https://store.printm3d.com/pages/support.

If you'd like to voice your input on improving our documentation, visit: 
- M3D's Official Troubleshooting & Self-Help Guide [Contribution & Suggestion Page](https://coda.io/d/M3D-Official-Troubleshooting-Self-Help-Guide_dzE73kMbIAL/Coda-Edits-Requests-Form_suKF6#_luq0Y)
- Promega Guides & Instructions [Contribution & Suggestion Page](https://promega.printm3d.com/#contributing)

-M3D Team

NOTE: This [sub-section](https://promega.printm3d.com/repair-and-maintenance/install-uninstall/slider) (within Promega Guide & Instructions) should help with installation. 

This [sub-section](https://coda.io/d/M3D-Official-Troubleshooting-Self-Help-Guide_dzE73kMbIAL/All-Printable-STL-Files_sugSV#_luTLj) (within M3D's Troubleshooting & Self Help) house the printed parts STL files.

closed 2019-04-03 by ericl-m3d