PrintM3D/Promega #27: Pro mega Network Setup

WilliamLalum, 2018-10-14:
We are unable to get either of our laptops to connect to the pro mega.  We have changed none of the default settings on the sd card. 
We have an Ethernet cable attached to the printer to our router.  Lights come on as instructed.  Power is on to the duet board.  When we open a browser in chrome with the address of promega/ we get a IP address cannot be found. Any advice?

WilliamLalum, 2018-10-14:
Ok. We found the IP address the router is assigning to the printer. Should we set the M552 command to that address on the SD card?

sohjsolwin, 2018-11-01:
> The instructions assume that your router has a built-in LAN DNS for devices that use DHCP to get their IP addresses. It's a bad assumption because most low cost routers don't.

I disagree with that statement. Every _router_ I've ever owned, even the cheap ones, have assigned IP addressed to devices connected to them. _switches_ on the other hand, just pass the connection along and do not assign IPs. 

> Ok. We found the IP address the router is assigning to the printer. Should we set the M552 command to that address on the SD card?

No, you do not need to set the IP in the SD card, unless you want to use a static IP. If that is the case, you should also make it static on your router so that on a renew it does not potentially assign that same IP to something else on your network first. If you do not assign a static IP, you should be able to access the printer from http://promega/ regardless of which IP the router ends up assigning to it.

TheBum, 2018-11-01:
I’m referring to the instructions that say to type “promega” into the address field in the browser. That doesn’t work on most home routers.

ericl-m3d, 2019-04-03:
Please log ONLY issues pertinent to software bugs or operational anomalies (suspected to be software-related). Any other type of issues logged will be not be addressed and will be deleted.

If you are looking for support, feel free to look through our ever-improving documentation for quick help.

- https://coda.io/d/M3D-Official-Troubleshooting-Self-Help-Guide_dzE73kMbIAL/Introduction_sucuq#_lulpw
- https://promega.printm3d.com/

Our support ticket is available at https://store.printm3d.com/pages/support.

-M3D Team

closed 2019-04-03 by ericl-m3d