PrintM3D/Promega #29: Sensor on compound head shows 25 degrees cold offset

sanslos, 2018-10-17:
The idle (cold) temperature on the Compound head shows around 0C.  It heats and cools, but starts and stops around 0C.
[image]
[image]
[image]

sohjsolwin, 2018-11-01:
what version of the duet firmware are you on? Have you updated it from what the printer originally shipped with? There was a change in one of the beta versions that modified the temperature readings a bit. I believe that may be related to what you're seeing (only if you've updated the firmware though).

sanslos, 2018-11-01:
It's a little more complicated than that.  This problem (low temp on Compound) was first seen with the original firmware.  I upgraded the firmware (currently 2.02RC3), no change, re-calibrated, no change.  I have a similar problem with K'Tana, where initially both sensors read about the same (around 27C at idle), then one sensor suddenly dropped to showing about 12C at idle (no change in firmware).  Note that I suspect a firmware issue, but I put a new sensor in the Compound head and it showed 27C again.  Something weird is going on.
[image]

![k tana initial bringup temps](https://user-images.githubusercontent.com/43072647/47858
[image]
183-aa86b800-dda8-11e8-8dea-90e085757903.png)

sanslos, 2018-11-01:
(Sorry, thought this posted)  This is what the K'Tana shows when first turned on

[image]

sanslos, 2018-11-01:
The lower temperature is consistent, it doesn't bounce around.  T1 now always shows 12C when idle.

sohjsolwin, 2018-11-01:
definitely strange. Have you run a PID tuning yet? https://promega.printm3d.com/repair-and-maintenance/temperature-calibration

sanslos, 2018-11-01:
Yes, I have (for the Compound).  I haven't done it for the current problem, I was planning to, but that wouldn't explain why it started out with both sensors showing 27C then one suddenly dropping to 12C.

sanslos, 2018-11-06:
The problems I'm seeing are due to the sensors.  I did experiments with the four sensors (Compound, 2 from K'Tana, one replacement) and two Duet ports, I currently have two failed sensors that show low.  Note the original Compound sensor now shows -18C, they deteriorate over time:

                          Duet Port (C)
Sensor   Ohms     Left    Right
        1    1046    12.40    12.00
        2    1106    27.90    27.70
        3    926    -18.10    -18.30
        4    1107    28.00    28.00

        Note: Wire Resistance = 0.6 ohms

c-schmitz, 2019-05-03:
I can confirm similar issues. Both my sensors are off by -15°C now. In the beginning they worked normally. 
This caused the head being too hot at nominal temperature.

I was able to temporarily fix that by messing with the R value in the configuration However this seems to be a general quality problem with the sensors (like so many other things)

closed 2019-09-27 by michaelarmani

c-schmitz, 2019-09-29:
Why was this closed?

sanslos, 2019-09-29:
The issue was with the sensors, when they were replaced the issue was fixed.  I should have closed this when that was verified (last November, 2018), it was the point of my Nov. 6, 2018 post saying "The problems I'm seeing are due to the sensors", although I may not have been clear.

Note that at the time I was also having problems with the heating, [I posted a whole series of experiments on Discord](https://discordapp.com/channels/279255051036524544/461926470776782848/509255194782072852).  Once I replace the sensors and the suspect heaters everything worked fine.

For completeness I should mention that at least once (twice?) I was not able to keep temperature, which was a Duet issue.  A hard power cycle (unplugging the printer) solved that.