PrintM3D/Promega #30: Grinding noises during printing....

MSI-Promega, 2018-10-19:
New user. Have my printer running, but we hear grinding and squeaking noises and I have yet to isolate their location. Perhaps from the belt guide rollers...? Has anyone been experiencing this as well?

jbolene, 2018-10-19:
yes I have the noises also

tirelessirony, 2018-10-19:
I also have heard grinding noises. I picked up some compressed air and try to blow out the front and side idler pulleys every once in awhile. I think small bits of worn belt, dust, debris can get into the spaces between the smooth bearings and cause surprisingly loud grinding. 

That doesn't completely solve it, but between that and cleaning/lubricating rods and rails - I've heard less. I use PTFE lube on the rails and dry-film teflon on the rods.

bass4aj, 2018-10-19:
I have encountered squeaking and grinding from 2 places: 1. is the front
idlers/bearings. The debris from the belt gets in the bearings. I just
replaced them with GT2 Idlers since they are more protected, but others
have soaked their bearings in kerosene and re lubricated them.
2. is the PTFE tubing for the filament. If it rubs along the x-gantry long
enough, it will start to screech.

closed 2019-04-03 by ericl-m3d

ericl-m3d, 2019-04-03:
Please ONLY log issues pertinent to software bugs or operational anomalies (suspected to be software-related).

Any other issues logged will be not be addressed and will be deleted.

- M3D Team