PrintM3D/Promega #33: Print Cooling Fans Popped and Smoked

talonzme, 2018-11-18:
Anyone know the specs on the print cooling fans (2 little ones inside shroud)?  I was running a print yesterday and they popped then started smoking (badly).  I had to unplug them.  They won't run now,obviously, so I will need to replace them.  Not sure is the printer warranty is still in effect or not.  No changing of wires was done since the voltages are different between these and the nozzle fan.  I cannot find specs to replace the two print cooling fans though.

TheBum, 2018-11-18:
Since both blew out at the same time, I would suspect the Maestro board. In any case, it absolutely should be covered by the warranty.

talonzme, 2018-11-19:
How do I start the process to get them replaced? I am not seeing any other place to submit an issue but here....

jbolene, 2018-11-19:
https://store.printm3d.com/pages/support

talonzme, 2018-11-19:
Thank you jbolene. I realized I didn't have an account for that page. I have submitted the "chat" to request the fans.

closed 2019-01-02 by ericl-m3d