PrintM3D/Promega #51: I have no idea how to use this Promega

obeydesign, 2019-06-17:
BUT I actually got it to commune with my network. However, if I understand the machine at all (not saying I do) the two feeder tubes that go through the plastic chain are not attached to anything. To what should they be inserted into.

ALSO there is a gray piece at the back of the extruder (just out of the box). Is this something to be removed? It has a few screws, and maybe THAT'S where I'm supposed to insert the filament feeding tubes? Maybe?

I'd appreciate any help. I find the online manual to be nearly worthless, especially since the included pictures are so small.

TheBum, 2019-06-17:
That piece at the back of the extruder is what corrals the connections in the head. It needs to stay in place.

TheBum, 2019-06-17:
As for feeding the filament, a lot of people don’t use those feed tubes because they provide an unnatural sharp turn into the extruders. Not using them does require that the top window on the printer be left off.

If you take a flashlight and look down into the gaps in the top of the extruder, you can see the gear that grabs the filament. Insert the filament into that path with the nozzle heated up and issue extrude commands through the web interface to feed it in.

obeydesign, 2019-06-17:
Hey, thanks for writing back. I greatly appreciate it. I'll see if I can find the gaps you're referring to. Also, did you get a piece of glass to fit on the print bed? I got four large pieces of plastic, obviously for the top and sides, but I don't think I can use plastic as the print bed.

obeydesign, 2019-06-17:
[image]

This is the top of my extruder. Isn't it?

TheBum, 2019-06-17:
I’ve circled the filament entry locations.

https://www.dropbox.com/s/891w4ugq1dmihe9/IMG_0272.jpg?dl=0

obeydesign, 2019-06-17:
[image]
Thanks! Does this look correct? I've fed it down as far as it will go. Does that mean...it's in?

TheBum, 2019-06-17:
Farther forward. The filament needs to go in front of the plate that the last few threads of the screw is going through.

obeydesign, 2019-06-17:
[image]
Like this?

TheBum, 2019-06-17:
That’s the right hole, but I can’t really see the angle. What happens when you try extruding? Make sure you heat up the nozzle first.

obeydesign, 2019-06-17:
Oh, I gotta be honest, I'm waiting for a flurry of answers to many other questions from M3D. I've been using Octoprint, and this whole Duet board is all new to me.

Thank you again for your time. I greatly appreciate it.

obeydesign, 2019-06-18:
Okay. I got the filaments inserted. The extruder works so far. I even went out and got a $10 piece of class to fit over the heated bed. NOW I can't get the bed to Z-height adjust. I use the instructions, but...well, like I said, no joy.