PrintM3D/Promega #53: Tuning home position: Z-0.5 is maximum

wattronix, 2019-09-27:
I am trying to set Z home for the first time. 
As I instruct the bed rise 0.5mm at a step using the Machine Control of DWC, once I try to go from Z-0.5 to Z-0.55 or Z-0.6, the head refuses to go up such that I can slowly find the level where the nozzle touches the glass plate.

Suggestions as to how to overcome are welcomed.

michaelarmani, 2019-09-27:
There is a small amount of backlash due to tension in the belts and gravity that has to be considered.  In a fine tuned system, the belts are all tight, the pulleys are checked that they are flat on the shaft and tightened,  the Z sliders on the rails are lubricated and smooth to give low friction, and extra weight is added to the bed such as thick glass to ensure gravity helps absorb backlash.  Even in such a system there will be no less than Z = 0.10mm backlash.  Please identify your backlash (it could be in the range 0.1-0.3mm usually).  From here, use this when doing calibrations, or Z-hop during prints.

wattronix, 2019-09-28:
Hello Michael,

Thank you for the response. I measured the shortfall with some feeler gauges and inputted the value.

I moved on to Set up the bed probe and assumed that the statement “Engage bed probe” was the title of the photo below. It was an in reality a not well placed instruction to lower the bed probe.
Not knowing what the G30 referred to, the bed forced its way into the nozzle and jumped down about 10mm.

Realizing the error, I homed the bed and restarted. Going thru the correct procedure with the probe lowered when required, the bed now stops quite far (~5mm) away from contacting the nozzle.
The bed Z position comes as high as Z-0.5 and will not rise further when I activate the Move Z buttons, as instructions dictate, to move the bed up to contact the nozzle. I should be having a reading for Z at this step to about -7.0.

Hoping you can guide me further.

Ron

michaelarmani, 2019-09-29:
You are correct, Engage probe was meant to indicate the user manually deploys the probe.  

By hitting the nozzle, you've got the machine thinking it is inside the nozzle, then when it fell back ~10mm it thinks it is way off course - there is a "software endstop" that protects the nozzle from Crashing into the bed once calibrated.

To bypass this, try your procedure again but first put in this gcode command into the web interface:  M564 S0 H0, or you should be able to restart the printer (power cycle the blue-lighted switch for about 5 seconds off then on).

At the stage you are at I think you would be best served looking at the Duet documentation and talking amongst the other users in the discord or with our tech support team as the Github should be used to improve the code set if there is a systemic issue.    Please visit https://discord.gg/eQ5wPwk for the community or http://support.printm3d.com/

wattronix, 2019-10-09:
Oct8/19 (9pm)
Hello Michael,

Taking your recommendation, I took my Cura question to discord on Sept. 29
- No response from anyone to the "wattronix" posting to date.
*********************************************************************

Pressing on, I ignored the need to input values per the Promega instruction sheets, and
managed to get one side of the compound extruder to print a small object.The other side 
of the extruder did not work so well. 

I wrote to M#D support on Oct 6 and am still waiting anxiously. I wrote:
 "I was able to print my first model today using the right hand extruder.
The left extruder will not retract or extrude.
Motor 1 & 2 appear to be wired correctly to the DUET board.

With wiring as received, "Mixing as single Left"- No retract, no extrude. Will not print.
"Mixing as single Right"- Retract OK,  Extrude OK. Prints fine.

When motor connectors are reversed on Duet board, 
"Mixing as single Left"- retract is reversed, extrude is reversed and 
filament loads ~1" and then skips at feedrate of 5mm/sec. Extracts ~1" OK.
"Mixing as single Right"- No retract, no extrude. Will not print.

After the above test, 
1. the 2 motors seem to be operational.
2. the output from Duet board transfers "no functionality" from faulty left side 
	to right side extruder.
3. the left extruder skips at 5mm/sec extrusion rate.

I would like your analysis of why the left extruder is non-operational plus steps 
	to be taken to rectify.
I would like your explanation as to skipping of the left extruder at 5mm/sec rate.
I would like instructions as to how to remove and re-attach the gray extruded fan 
cover on sides and front of extruder. (I would like to be able to see if/when any 
motor gear jam takes place.)

Regards,"

************************
Michael, I am hoping that you can spur at least a response from M3D Support.
Regards,
Ron

ftcros, 2019-10-23:
When I was first setting the Promega up I frequently had lag after issuing a command. I then got a screen popup indicating 'connection lost - wait for re-connection' meaning the network IP had lost its connection with the printer. I'm still trying to resolve this problem and have tried a 'static' ip with and without MAC code (\sys\M540 in machine_access.g) 

M540 PBE:EF:DE:AD:FE:ED     ; Set MAC address, this can be used to assign a given IP in the DHCP
M552 P0.0.0.0 S1             ; Use this to enable DHCP
; M552 P192.168.1.122 S1        ; Set Static IP address and enable networking

(; M552 P192.168.1.122 is the IP assigned by my DHCP so I incorporated it in the static IP - but still have times when 'connection lost' appears). At present I'm using DHCP with MAC address but I'm also considering using a USB/RJ45 3.0 adapter to connect the printer directly to my PC.