PrintM3D/Promega #54: M3D Promega Print bed unstable

ftcros, 2019-10-23:
I am (1) unable to mesh the print bed with results better than very poor (a lot of red on the graph).  I'm using what was shipped with the printer. My print bed is a green, thin 'fiberglass' like material. Should I have a glass print bed?

michaelarmani, 2019-10-23:
The green sheet is an "unbound" print surface made by us.  This is the preferred material and what we use in the factory.  Others have found that buying a very thick marble or glass sheet at their local store is a little more flat.  We also sell a medium thickness glass sheet.  "A lot of red" doesn't mean anything itself as you may just have the bed probe offset set wrong.  what matters is the range and RMS deviations.

ftcros, 2019-10-24:
Thanks, once I applied enough Elmers Glue I began to get results. 
The 'red' has become less of concern after I released and then reapplied the bed clips. The bed was brand new when I first secured it in place and began to bulge when heated - now it's more 'seasoned' and lays flatter. Thanks

closed 2019-10-24 by ftcros