PrintM3D/Promega #59: M3D Promega filament feed produces next to nothing or sqiggles

ftcros, 2020-11-12:
I have not, to date, produced more than one print and it was not what I intended. 
I've read a few users are not getting anywhere with filament feed. I've used printhead heating methods and slower rates to no end. 
Please feel free to enlighten me, please.

marcel-isler, 2020-11-14:
After spending much time trying to beat the stock setup into submission, I ended up redesigning most of the printer and I'm very happy with how it prints now.  Look on Facebook for the ShadowMega Support Group.

Biothermo, 2020-11-25:
I've struggled too with the default configuration. I've had some success, but recently there have been problems with insufficient filament creating a lot of voids.  I noticed a lot of popping noise from the printer head when the problems are worse.  Following these [suggestions ](https://3dprinterly.com/8-ways-how-to-fix-a-clicking-slipping-extruder-on-a-3d-printer/)helped, but didn't solve the problem. Next I disassembled the head and found the [Single K'tana Plastic Insert ](https://promega.printm3d.com/documentation/mechanical-systems/single-ktana)to be badly worn.  This is a 3D printed part which I have [downloaded](https://promega.printm3d.com/printed-parts-bin/bin/extruder-system), but not yet printed. 

This is a high-wear point, not suitable for the soft plastic that was provided.  Now I'm wondering what would be the best replacement.

Good luck, Biothermo

ftcros, 2020-11-25:
By no means perfect but the original print head produced for the first time. Had I not found a slicer program, https://slic3r.org/, and slewed the print rate back 50%, it might never have happened. 

Reply to ftcros@ftcrosby.com